LibreSolar/charge-controller-firmware #38: Firmware does not work unless UART_SERIAL_ENABLED is defined in config.h

hogthrob, 2019-06-27:
I found the cause for this to be sleep().  If UART_SERIAL_ENABLED is not defined and subsequently the ISR is not attached, it does not return, so after 10s the watchdog gets triggered and restarts the devices. 
I have no idea why sleep returns in a timely manner if we have the serial port rx interrupt attached, but it is what I found out.

hogthrob, 2019-06-29:
I found that the reason why UART_SERIAL_ENABLED keeps the whole thing working even when calling sleep is that when the receive interrupt is attached, the mbed code calls `sleep_manager_lock_deep_sleep();`
preventing deep sleep. And we do the same in the main() function, sleep works also for all other configurations. Essentially this controls which sleep mode is entered. And at least for the STM32F072 the deep sleep mode does not work as intended (or is not configured correctly).

I have no idea if this is working in the L073. 

BTW, the branch https://github.com/hogthrob/MPPT-Charger_Software/tree/thingset_device has that fix and also the reworked thingset device concept with classes and objects.

martinjaeger, 2019-06-29:
Cool, I'll have a closer look during the next week.

hogthrob, 2019-08-31:
This issue was fixed by #46 , https://github.com/LibreSolar/charge-controller-firmware/blob/7e1d1374a2b998ae24baea07d5877fbff2abd9af/src/main.cpp#L168

closed 2019-08-31 by hogthrob