LibreSolar/charge-controller-firmware #43: charger_detect_num_batteries returns always 12V battery

Dee-Fuse, 2019-07-26:
I'm trying to set the MTTP 2420 with an 24V battery but the controller reports only 12V battery.

I think the multiplication by 2 here https://github.com/LibreSolar/charge-controller-firmware/blob/3261414fcf0b6461d44af96b37778d95383ab4c2/src/bat_charger.cpp#L221-L222 is incorrect.

If I set `BATTERY_NUM_CELLS` in "config.h" to 12 as suggested for 24V systems the if statement is always false and defaults to 12V (actual values added in [brackets])
```
    if (bus->voltage [24.4] > bat->voltage_absolute_min * 2 [38.40] &&
        bus->voltage [24.4] < bat->voltage_absolute_max * 2 [58.80]) {
        charger->num_batteries = 2;
        printf("Detected 24V battery\n");
    }
    else {
        printf("Detected 12V battery\n");
    }
```

martinjaeger, 2019-07-26:
Well, you're right. It's a bit misleading. Actually the settings should work fine, but the software assumes that one battery is a 12 V battery (6 cells lead acid or 4 cells LFP). If you configure 6 instead of 12, it should also print the setting correctly.

Dee-Fuse, 2019-07-27:
Wouldn't that alter the charge controllers target voltages?
the variable `charger->num_batteries` has only a direct effect on the variable `bus->chg_voltage_targe` and influences some `if` statements, whereas the number of cells alters almos all battery parameters for the statemachine control.

martinjaeger, 2019-08-08:
Ok, so I will change the message to make it more clear and close the issue afterwards.

Please reopen if you disagree.

commit 0a2a3869 (2019-08-08): 12V/24V battery detection message updated (closes #43)
closed 2019-08-08 by martinjaeger