LibreSolar/charge-controller-firmware #44: MPPT does not work if connected load is higher than power provided by panel

hogthrob, 2019-08-25:
I noticed that a couple of times that the MPPT 2420 turned off using power from the panel in full sunshine which is strange. 
It turns out that in line 
https://github.com/LibreSolar/charge-controller-firmware/blob/3f4f52ea18ab81bd18cd69d7960e3a3e7b6bac76/src/dcdc.cpp#L52

the battery charging current is used to calculate the supplied power. Which goes to negative if the connected load is drawing more power than the panel is able to supply. This forces the MPPT try to get even more power, which basically increases the PWM duty cycle to max duty cycle, and then panel voltage drops. 
The fix for me was to replace "out->current" in this line with "dc->ls_current" which represents the actual power provided by the panel. Now it seems to work as expected. However, I haven't looked into all possible cases (like boost charging etc). 

Maybe my fix is not suitable in all cases, but the issue as such exists.

martinjaeger, 2019-08-25:
Wow, very good detective work :)

I totally agree that this is an error. In addition to the case where load power > DC/DC power, the MPPT algorithm could also get disturbed when the load power changes.

Not 100% sure how to fix in the best way though. The `_dcdc_output_control` function works independent of boost or buck mode. So if we take `out->voltage * dcdc->ls_current`, the boost mode won't work properly anymore.

I see two options:
1. Store also the `ls_voltage` in the `Dcdc` struct and take the absolute value of the `ls_current`.
2. Pass some additional parameters to `_dcdc_output_control`.

Probably option 1 is better. What do you think?

In general, we don't really need the option `dcdc->power` anywhere else than in this function, so we could also store the previous value as a static variable.

hogthrob, 2019-08-25:
I also think 1) is the cleanest solution. We collect and store the information about the dcdc power together in a single place. It is a small change to adc_dma.cpp and dcdc.cpp. 
While agree that dcdc->power is only used in a single place, I rather tend to keep it. 
It is a property of the dcdc algorithm / object and should be located there. I am not a friend of static variables in functions to be honest. Although in a scenario like this, where the DCDC object is a singleton, it doesn't make much of a difference.

martinjaeger, 2019-08-25:
Yes, makes sense to keep it in the `Dcdc` struct. So we go for solution 1.

Should I implement the fix (probably tomorrow) or do you want to send a PR?

closed 2019-08-26 by martinjaeger