LibreSolar/charge-controller-firmware #50: Support load by solar power if battery is discharging but not discharged "enough"

hogthrob, 2019-10-07:
I recently added some code which keeps the solar panel delivering energy to the load even if the LiIon battery is fully charged (we simply keep the charge current as close as possible below 0). Now with an increased battery capacity I came across another "issue": This reduced discharge by using the panel to power the load (but not charging the battery) works fine until the night and the next morning comes: The solar power will not be used even if it is available until we discharge under the restart charge voltage limit. 
This is cause by the check for the dcdc power transfer start, it checks if the charging is permitted (which is not, since we have not yet fallen below the limit):
https://github.com/LibreSolar/charge-controller-firmware/blob/987a6d8894f91dc2f930b79eec531a4b551295b6/src/dcdc.cpp#L125

I think we can drop this line altogether. In case charging is not permitted, we should keep the charge current below zero (i.e. no transfer into the battery), but otherwise, why not transfer energy? 
I opened this issue to discuss this. I would like to understand if there is a good argument for not removing this line. I know the change would work in my setup.

martinjaeger, 2019-10-07:
The idea was to have a generic DC/DC firmware module that can be used for buck and boost operation. If we remove the `out->chg_allowed` check, the DC/DC might possibly start to transfer power into the solar panel. This would currently be prevented because `chg_voltage_target` and `chg_voltage_min` are initialized with 0, so the `_dcdc_check_start_conditions` function would still return false. But it's maybe not very clean to completely ignore the `chg_allowed`.

I've got another idea: What if we remove `bool chg_allowed` and `bool dis_allowed` in the `DcBus` struct and introduce `float chg_current_target` and `float dis_current_target` instead. These could be set to 0 if no current in that direction is allowed. And the DC/DC would be controlled such that it tries to stay within the target range. If it's not possible to stay within the limit anymore (e.g. because the solar power drops too much and the solar current goes below `dis_current_target = 0`), we stop.

However, this alone would not yet solve the problem with the load. I realize there is a more general issue in the firmware at the moment. Let's take this picture to explain:
[image]

We essentially have 4 different current flows:
1. Solar current, stored in `hv_bus` and calculated from inductor current and voltage ratio,
2. Low-voltage side of the DC/DC converter (= inductor current), which is currently (erronously) considered the same as `bat_bus`.
3. Battery current, stored in `bat_bus`, which points to `lv_bus`
4. Load current, stored in `load_bus`

In the DC/DC control, we use both the inductor current (stored as `ls_current` in the `Dcdc` struct) and the output bus current (either high or low voltage side). This creates issues, as they are not the same. Essentially, I think the DC/DC controller should not see the battery current at all, but only the current in location `LV` as in the image. So we need a new `lv_bus` object, that is separate from the `bat_bus` object. If the battery is full and the load draws 2A, the `lv_bus.chg_current_target` would still be set to 2A and the DC/DC should start up properly. 

What do you think?

MatthiasN73, 2019-10-07:
This is an automatic reply to your message
 
 ---------------------------------------------------------------------
 Sie haben eine E-Mail an die Adresse m.neubauer@schenk-ansorge.de geschickt.
 Leider bin ich bis einschließlich 11.10.2019 18:00  nicht im Labor. 
 In dringenden Fällen wenden Sie sich bitte an Herrn Gerhard unter
 der Telefonnummer 0391/24468-246 bzw. F.Gerhard@schenk-ansorge.de

 Vielen Dank für Ihr Verständnis.

 ---------------------------------------------------------------------

 File names:            
 
I recently added some code which keeps the solar panel delivering energy to the load even if the LiIon battery is fully charged (we simply keep the charge current as close as possible below 0). Now with an increased battery capacity I came across another "issue": This reduced discharge by using the panel to power the load (but not charging the battery) works fine until the night and the next morning comes: The solar power will not be used even if it is available until we discharge under the restart charge voltage limit. 
This is cause by the check for the dcdc power transfer start, it checks if the charging is permitted (which is not, since we have not yet fallen below the limit):
https://github.com/LibreSolar/charge-controller-firmware/blob/987a6d8894f91dc2f930b79eec531a4b551295b6/src/dcdc.cpp#L125

I think we can drop this line altogether. In case charging is not permitted, we should keep the charge current below zero (i.e. no transfer into the battery), but otherwise, why not transfer energy? 
I opened this issue to discuss this. I would like to understand if there is a good argument for not removing this line. I know the change would work in my setup.

MatthiasN73, 2019-10-07:
This is an automatic reply to your message
 
 ---------------------------------------------------------------------
 Sie haben eine E-Mail an die Adresse m.neubauer@schenk-ansorge.de geschickt.
 Leider bin ich bis einschließlich 11.10.2019 18:00  nicht im Labor. 
 In dringenden Fällen wenden Sie sich bitte an Herrn Gerhard unter
 der Telefonnummer 0391/24468-246 bzw. F.Gerhard@schenk-ansorge.de

 Vielen Dank für Ihr Verständnis.

 ---------------------------------------------------------------------

 File names:            
 
The idea was to have a generic DC/DC firmware module that can be used for buck and boost operation. If we remove the `out->chg_allowed` check, the DC/DC might possibly start to transfer power into the solar panel. This would currently be prevented because `chg_voltage_target` and `chg_voltage_min` are initialized with 0, so the `_dcdc_check_start_conditions` function would still return false. But it's maybe not very clean to completely ignore the `chg_allowed`.

I've got another idea: What if we remove `bool chg_allowed` and `bool dis_allowed` in the `DcBus` struct and introduce `float chg_current_target` and `float dis_current_target` instead. These could be set to 0 if no current in that direction is allowed. And the DC/DC would be controlled such that it tries to stay within the target range. If it's not possible to stay within the limit anymore (e.g. because the solar power drops too much and the solar current goes below `dis_current_target = 0`), we stop.

However, this alone would not yet solve the problem with the load. I realize there is a more general issue in the firmware at the moment. Let's take this picture to explain:
[image]

We essentially have 4 different current flows:
1. Solar current, stored in `hv_bus` and calculated from inductor current and voltage ratio,
2. Low-voltage side of the DC/DC converter (= inductor current), which is currently (erronously) considered the same as `bat_bus`.
3. Battery current, stored in `bat_bus`, which points to `lv_bus`
4. Load current, stored in `load_bus`

In the DC/DC control, we use both the inductor current (stored as `ls_current` in the `Dcdc` struct) and the output bus current (either high or low voltage side). This creates issues, as they are not the same. Essentially, I think the DC/DC controller should not see the battery current at all, but only the current in location `LV` as in the image. So we need a new `lv_bus` object, that is separate from the `bat_bus` object. If the battery is full and the load draws 2A, the `lv_bus.chg_current_target` would still be set to 2A and the DC/DC should start up properly. 

What do you think?

hogthrob, 2019-10-08:
I understand the argument. 

However, since we are obeying the permitted charge current on the solar panel bus chg_current_max (which is should be set to 0 for a solar panel in dc_bus_init_solar() ), it will prevent charging a solar panel in boost mode scenarios while still supply energy to the load (which is on the same side as the panel in this case). 
In buck mode this will never happen as we are not switching in and out anymore.

BTW,  I am not against cleaning the code up. It took me quite while to wrap my head around this stuff in the dcdc.cpp area. Especially the trick with switching the in and out bus if we go to boost mode makes this very difficult. Making this cleaner is a good idea.

hogthrob, 2019-10-08:
We have in fact 3 supported configurations which we have to consider:
HV=Panel, LV=Bat+Load
HV=Grid, LV=Bat+Load
HV=Bat, LV=Panel+Load

martinjaeger, 2019-10-08:
I agree that the `dcdc` and `bat_charger` part got a bit complicated. But I didn't want to have different code for boost and buck applications. This is also why I tried to avoid using names like "solar" for the HV side.

Yes, you're right. We have these 3 configurations. However, I didn't use the load output when operating in boost mode.

martinjaeger, 2019-10-08:
What if we rename the different DcBus objects like this, with `lv_bus_int` as our new junction, which we use for DC/DC control:
```
DcBus hv_terminal = {};         // high voltage side of DC/DC converter (solar input for typical MPPT)
DcBus lv_bus_int = {};          // internal low voltage side of DC/DC converter
DcBus lv_terminal = {};         // externally accessible low voltage bus (battery for typical MPPT)
DcBus load_terminal = {};       // load terminal
DcBus *bat_terminal = NULL;     // pointer to terminal where battery is connected
DcBus *solar_terminal = NULL;   // pointer to above terminal where solar panel is connected
```

hogthrob, 2019-10-08:
In the grid case we have no solar terminal but a terminal which can be both a energy producer or consumer.  

I think, it would be indeed helpful if we separate the DCDC part into an kind of independent object, which has 2 operating modes only: energy transfer ON / OFF.

And some management code which, based on the current setup and operating conditions calculates the DCDC objects operation parameters and controls transfer ON / OFF.

The DCDC object should take care about remaining in the operational limits (such as max amperage, detecting of fault conditions like to high temperature of the related components) and reporting accordingly to the management code.

I hope is goes along your lines of thinking as well.

martinjaeger, 2019-10-09:
Yes, looks like we think towards a similar direction.

I would even say that the energy transfer on/off setting is not set externally, but the DC/DC object decides itself if it should be on or off. Just the boundary conditions are set externally (updated regularly based on measurements etc.).

I will work on the DC/DC part today and provide a suggestion. And I'm starting with the unit-tests, which were still missing for the DC/DC module.

martinjaeger, 2019-10-12:
It took a bit longer than initially planned, but here is a suggestion for improvement. Let me know what you think.

Still have to test it in hardware beginning of next week. But I wrote the unit-tests first, checked with old implementation and then rewrote the control functions. So I hope it works as intended.

martinjaeger, 2019-10-15:
During test in hardware I found out that I had to add a function to calculate the properties of `lv_bus_int` based on the load and battery settings. That's now implemented and everything seems to work as expected. Merged it to master.

Let me know if it also works on your end to close the issue.

martinjaeger, 2019-10-16:
Just identified a fallacy on my side. If the current of a `dc_bus` is always the sum of in-going (charging) current minus out-going (discharging) current, internal bus current can only be zero. But the actual current assigned to `lv_bus_int` in `update_measurements` function is equal to the DC/DC inductor current. The firmware works fine at the moment, but seems like the introduction of the internal DC bus failed to make things more clear... will re-think it again.

Probably current measurements need to be stored in the transferring objects like the DC/DC or the load switch. The DC bus stores only the voltage.

hogthrob, 2019-10-16:
Well. The sum of incoming and outcoing currents in a node/junction adds up to 0. Seems we have identified Kirchhoffs Law again ;-) . 

What we really want is that the energy needs for battery charging plus load driving are as much as possible coming from the inductor side. With the additional constraint of being able to control the battery charging  current independent of the load current (Necessary to be able to stop energy storage in case of fully charged battery).

martinjaeger, 2019-10-20:
Ok, here is a new suggestion that I pushed to branch [playground](https://github.com/LibreSolar/charge-controller-firmware/tree/playground).

I finally changed more in the code structure than initially planned, but I'm quite happy with the final result now.

The firmware can now be easily configured for any charge controller variant: PWM and/or MPPT, boost or buck mode in case of MPPT, load output existing or not.

Summary of improvements in the firmware:

- Differentiate between a `DcBus` (red) and a `PowerPort` (blue). The DC bus is where the voltage is stored. Each power port is connected to exactly one DC bus. Current, power and energy measurements/calculations are stored in the power ports, as well as current related limits.

  Picture of the possible charge controller configurations and naming of DC buses and ports:

  [image]

- I decided to use a few more C++ features to make the code easier to read. The `Charger`, `Load`, `PwmSwitch` and `Dcdc` structs were converted into classes, which are associated to a `PowerPort` via the constructor. So there is no possibility to forget initialization of pointers, etc.

  Instead of `dcdc_control(&dcdc)` we can now just write `dcdc.control()`.

- The previous variables `bat_terminal` and `solar_terminal`, which were previously pointers which are initialized in main function are now C++ references. Advantage: The references are initialized outside the function and can thus be accessed in the data_objects.cpp via `extern` keyword. So also the data objects change automatically if e.g. the battery is connected to the high voltage side in case of a boost converter.

hogthrob, 2019-10-21:
To use C++ where it makes obviously sense, is a good idea indeed. And the use of references instead of pointers:  👍 That solves often a lot of issues you may have with pointers.

I haven't look into the code in depth but as this is for sure moving into the right direction, let's go for it.

Just one question: I know it does not make much sense but it seems to me that the MPPT hardware could also implement a PWM charger if instead of a load a solar panel is connected to the load?
Which would enable a nanogrid case where the local battery can be charged from the grid and from a "local" solar panel.

martinjaeger, 2019-10-25:
Closing this issue as it should be solved with mentioned PR. If not, we can re-open it.

closed 2019-10-25 by martinjaeger

hogthrob, 2019-10-30:
Verified that #62 indeed seems to fix this issue. Thank you!

martinjaeger, 2019-10-30:
Great, that's good to hear. Thank you for testing and double-checking :)