LibreSolar/charge-controller-firmware #51: Firmware doesn't compile with most recent mbed version

martinjaeger, 2019-10-07:
The firmware does not compile anymore if the most recent version of mbed is used by removing the version specifier here: https://github.com/LibreSolar/charge-controller-firmware/blob/987a6d8894f91dc2f930b79eec531a4b551295b6/platformio.ini#L23

Error message:
```
src/thingset_device.cpp:44:5: error: invalid use of template-name 'ThingSetSerial' without an argument list
     ThingSetSerial ts_uart(serial, pub_channel_serial);
     ^~~~~~~~~~~~~~
src/thingset_device.cpp:44:5: note: class template argument deduction is only available with -std=c++1z or -std=gnu++1z
In file included from src/thingset_device.cpp:40:0:
src/thingset_serial.h:55:28: note: 'template<class T> class ThingSetSerial' declared here
 template<typename T> class ThingSetSerial: public ThingSetStream
                            ^~~~~~~~~~~~~~
src/thingset_device.cpp:66:6: error: 'ts_uart' was not declared in this scope
     &ts_uart,
      ^~~~~~~
src/thingset_device.cpp:74:1: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<ThingSetDevice*>'
 };
 ^
*** [.pio/build/mppt-1210-hus-v0.7/src/thingset_device.o] Error 1
cc1: warning: command line option '-Wno-register' is valid for C++/ObjC++ but not for C
```

@hogthrob : Do you have an idea? Unfortunately, my C++ knowledge is too little to find a fix immediately.

MatthiasN73, 2019-10-07:
This is an automatic reply to your message
 
 ---------------------------------------------------------------------
 Sie haben eine E-Mail an die Adresse m.neubauer@schenk-ansorge.de geschickt.
 Leider bin ich bis einschließlich 11.10.2019 18:00  nicht im Labor. 
 In dringenden Fällen wenden Sie sich bitte an Herrn Gerhard unter
 der Telefonnummer 0391/24468-246 bzw. F.Gerhard@schenk-ansorge.de

 Vielen Dank für Ihr Verständnis.

 ---------------------------------------------------------------------

 File names:            
 
The firmware does not compile anymore if the most recent version of mbed is used by removing the version specifier here: https://github.com/LibreSolar/charge-controller-firmware/blob/987a6d8894f91dc2f930b79eec531a4b551295b6/platformio.ini#L23

Error message:
```
src/thingset_device.cpp:44:5: error: invalid use of template-name 'ThingSetSerial' without an argument list
     ThingSetSerial ts_uart(serial, pub_channel_serial);
     ^~~~~~~~~~~~~~
src/thingset_device.cpp:44:5: note: class template argument deduction is only available with -std=c++1z or -std=gnu++1z
In file included from src/thingset_device.cpp:40:0:
src/thingset_serial.h:55:28: note: 'template<class T> class ThingSetSerial' declared here
 template<typename T> class ThingSetSerial: public ThingSetStream
                            ^~~~~~~~~~~~~~
src/thingset_device.cpp:66:6: error: 'ts_uart' was not declared in this scope
     &ts_uart,
      ^~~~~~~
src/thingset_device.cpp:74:1: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<ThingSetDevice*>'
 };
 ^
*** [.pio/build/mppt-1210-hus-v0.7/src/thingset_device.o] Error 1
cc1: warning: command line option '-Wno-register' is valid for C++/ObjC++ but not for C
```

@hogthrob : Do you have an idea? Unfortunately, my C++ knowledge is too little to find a fix immediately.

hogthrob, 2019-10-07:
I will have a look. Strange. This looks more like an issue in our code which is not changed by removing the version identifier. We'll see what this is about.

MatthiasN73, 2019-10-08:
This is an automatic reply to your message
 
 ---------------------------------------------------------------------
 Sie haben eine E-Mail an die Adresse m.neubauer@schenk-ansorge.de geschickt.
 Leider bin ich bis einschließlich 11.10.2019 18:00  nicht im Labor. 
 In dringenden Fällen wenden Sie sich bitte an Herrn Gerhard unter
 der Telefonnummer 0391/24468-246 bzw. F.Gerhard@schenk-ansorge.de

 Vielen Dank für Ihr Verständnis.

 ---------------------------------------------------------------------

 File names:            
 
I will have a look. Strange. This looks more like an issue in our code which is not changed by removing the version identifier. We'll see what this is about.

martinjaeger, 2019-10-08:
Fixed with #52

closed 2019-10-08 by martinjaeger