LibreSolar/charge-controller-firmware #69: Load + USB output state machine rework

martinjaeger, 2019-10-31:
Currently, USB and load output are both mangled in one file and have similar state machines. However, the state machines are not exactly the same, so we have lots of duplicated code.

In addition to that, errors in the load output sometimes need to overrule also the USB output. So we have duplicated code also in the error checking mechanisms.

The load output state is currently also used to indicate errors (saying why it is off and not only that it is off). Most of these errors are also indicated in the global `log_data.error_flags`. So we have another redundancy and source of issues here.

Don't have a good idea for improvement yet, but want to raise the issue so that we can think about it and discuss suggestions.

martinjaeger, 2019-11-09:
I'm currently working on this topic. Will also remove lots of duplicate code here, @hogthrob.

db4ple, 2019-11-09:
Appreciated 👍

martinjaeger, 2019-11-11:
Done for the load state machine in commit f254c77ab6793b393439f74d2e2ea31045fc450b.

USB state machine is still as before. We should probably also add USB-specific error flags, as the SMPS on the MPPT 1210 HUS has a power-good signal that can be read by the microcontroller.

martinjaeger, 2020-05-06:
USB state machine was also updated in the meantime. USB and Load output are now based on the same C++ class.

closed 2020-05-06 by martinjaeger