LibreSolar/charge-controller-firmware #70: Indication if solar panel is connected with reverse polarity

martinjaeger, 2019-10-31:
At least the PWM charge controller can measure the (negative) voltage of a solar panel connected with reverse polarity (if battery is already present). This could be indicated by a flashing LED. In addition to that, we could introduce an additional error flag for this.

Any suggestion for LED indication, @JordanSilverman?

I think it's nothing time critical, but nice to have for the future.

JordanSilverman, 2019-10-31:
My suggestion would be that we either continue using the load LED to
display faults and have a different blinking pattern for each error flag.
Or we can use the 'data' LED but still implement different patterns for
different errors. This LED isn't used very often but might not be as clear
that there is a fault.

The different patterns can be described in the manual so people can debug
on site.

Cloud Solar specific comment:
I also suggest that we do send the error flag over the communications (and
also add the hardware and firmware versions). We can perhaps add them
server side tomorrow and decide how to display them at a later date. Would
you agree?

martinjaeger, 2019-11-01:
I think we should use a different LED than the load LED, so that load errors and solar input errors can be distinguished.

The 5th (data / RX/TX) LED is a good idea. We could use fast blinking (flicker) as an error indicator in general. and then blink instead of flicker after we received an incoming message. Or we flicker the upper SOC indicator LED that is normally blinking if it is charging properly.

Yes, agree to store the error flag already on the server and add UI information later.