LibreSolar/charge-controller-firmware #79: Artificial current limit

hogthrob, 2019-11-09:
The line below effectively halfs the current limit for battery charging:

https://github.com/LibreSolar/charge-controller-firmware/blob/ba7d379225c2064b5b7a17177205bdbee5a6b26b/src/power_port.cpp#L76

Lets do the math:
```C
p_bat->pos_current_limit = 10; // max permitted charge current
p_bat->current  = 5; // current battery charging current
p_load->current = 0; // no load connected
=> p_dcdc->pos_current_limit = 5;
```
So far so good, but now have a look at these lines from the dcdc current control:
https://github.com/LibreSolar/charge-controller-firmware/blob/ba7d379225c2064b5b7a17177205bdbee5a6b26b/src/dcdc.cpp#L101-L106

Here we reduce the current once we are higher than p_dcdc->pos_current_limit 
which is 5 if we have 5 amps of charging current. So if we ever go over 5A the current is reduced to around 5A.
But the real limit is 10A, not 10A/2. With this code we will never reach it. I tried this with a lab power supply and the code real does limit the current in the way described above.

I think, the  formula should be  `p_bat->pos_current_limit + p_load->current`

martinjaeger, 2019-11-09:
Yes, I agree after thinking about it twice. The existing calculation is wrong. I must have had the remaining difference in mind, but the DC/DC is controlled to match the absolute current, of course.

I will check the charge controller that is in operation with this firmware. So far I didn't realize it doesn't get to full power. But that might be because the weather has not been ideal during the last few days so that a horizontally mounted 280W rooftop panel generates only 70-80W.

hogthrob, 2019-11-09:
I only noticed this using the power supply as solar panel where I can let the sun shine bright even in the darkest night...

martinjaeger, 2019-11-09:
Yep, that's good.

I just wonder why I didn't see this issue with the PWM charge controller. It uses the same function to calculate its output port maximum current and I have done several tests at full solar power with it in the last few days (also with a power supply). I will try to find out the reason...

martinjaeger, 2019-11-09:
Ah, the obvious reason is: I don't use very small batteries and also the standard capacity setting in the config is 40Ah. The default max. charge current is set to 1C:
https://github.com/LibreSolar/charge-controller-firmware/blob/ba7d379225c2064b5b7a17177205bdbee5a6b26b/src/bat_charger.cpp#L33

So we should see this issue under typical conditions only if the battery capacity is very low (and configured as such).

hogthrob, 2019-11-10:
It happens if the battery capacity in Ah is less than twice the rated hardware dcdc current in A and your solar input current exceeds half of the rated capacity in A, so with the 2420 for batteries with less than 40Ah, for 1210 less than 20 Ah.

martinjaeger, 2019-11-11:
Fix is now implemented.

closed 2019-11-11 by martinjaeger