LibreSolar/charge-controller-firmware #84: Dead link on this Github page

itofficeeu, 2019-12-27:
Sorry for coming up with such minor thing. - Maybe even out of scope?

On the page:
https://github.com/LibreSolar/charge-controller-firmware/blob/master/docs/firmware.md

is the link to
https://libre.solar/docs/toolchain

dead.

The link is in the words "develop software", on the section:
https://github.com/LibreSolar/charge-controller-firmware/blob/master/docs/firmware.md#toolchain-and-flashing-instructions

I encountered the problem because of my interest in working with the software. I was interested in the information. Is there a new page replacing the dead link?

Cheers,

martinjaeger, 2020-01-07:
Thanks for this comment. Well, I just realize that the firmware.md is somewhat outdated anyway.

The vanished page described some basics about how to use PlatformIO. But it was too little information to be actually useful, so I deleted it some time ago.

I'm planning to improve overall firmware documentation with some more graphs showing the structure. Will fix this dead link at the same time. So I'll keep this issue open till then.

martinjaeger, 2020-02-28:
Just updated the firmware docs a bit and removed the dead link. So I'm closing this issue now.

closed 2020-02-28 by martinjaeger