LibreSolar/charge-controller-firmware #92: Undervoltage Condition disables charging permanently

hogthrob, 2020-02-26:
What is the problem: 

If the battery was discharged until the load disconnects, charging will never commence.

Problem Analysis:
If the load gets disconnected, we finally set the ERR_BAT_UNDERVOLTAGE in Charger::discharge_control() here:
https://github.com/LibreSolar/charge-controller-firmware/blob/2154a1cb72612ce70c181a5432678fa22c61ed70/src/bat_charger.cpp#L303

Only place we clear the flag is this:
https://github.com/LibreSolar/charge-controller-firmware/blob/8539fa35572731f1cb6a8a48f7904743a2f6c33c/src/bat_charger.cpp#L336-L349

Since we will now never pass the start condition check (which tests for ERR_BAT_UNDERVOLTAGE) here:
https://github.com/LibreSolar/charge-controller-firmware/blob/2154a1cb72612ce70c181a5432678fa22c61ed70/src/dcdc.cpp#L132
we will never clear the flag ERR_BAT_UNDERVOLTAGE unless the battery voltage increases above the bat_conf->voltage_load_reconnect. However, how should the battery voltage magically increase above the limit without being charged? 

The underlying question is: What is the exact meaning of ERR_BAT_UNDERVOLTAGE?

1. It just indicates a too low battery voltage for load driving, we should have this stopping the charging start (remove check). 
2. It is meant to indicate a dangerous condition, i.e. we discharged below a critical point and don't want to start charging because this would possibly indicate a broken battery and thus charging could be dangerous, we should remove the ERR_BAT_UNDERVOLTAGE setting from bat_charger and only keep the remaining check in adc_dma for "real" undervoltage conditions.

I tend to interpretation 2.

martinjaeger, 2020-02-26:
Good observation. I even saw that a charge controller didn't continue charging even though the battery was not full, but thought it was because the device had a hardware fault (which it actually had).

I am working on the load output and issue #91 at the moment, so this error may have been solved already, but I'll double-check.

Originally, I wanted to push the changes to a separate branch `load_out_rework` and do some further testing before merging it to master, but for some reason `git push origin load_out_rework` also updated the master, where I added the commits locally already... as I don't want to force-push to master, I'll keep them in now.

What's missing is mainly the USB port testing with the new more generic `LoadOutput` class (inheriting from `PowerPort`). I'll try to do that and check aswell if above error is resolved.

With the update, the `ERR_BAT_UNDERVOLTAGE` is not used for load driving anymore. Instead, each load has its own voltage limit (allowing to set different thresholds for USB output and the actual load terminal). The flag is still existing in the charger, but currently not used anymore.

martinjaeger, 2020-08-18:
Just committed recent changes from develop branch which should have finally fixed this issue. Double-checked again today.

The switch to Zephyr RTOS is now almost finished, just some final testing missing until I'll create a new firmware release.

closed 2020-08-18 by martinjaeger