LibreSolar/charge-controller-firmware #116: Simple adc_get_value function non-existant

luizvilla, 2021-02-11:
When developing new functions from the already existing ones, it is very easy to get lost on the code. Simple functions, such as adc_get_value do not exist (or we have not been able to find them). Maybe a certain set of basic functions should be defined so that everyone can build their applications from them.

martinjaeger, 2021-02-11:
Yeah, I agree. Would be helpful for development and testing.

Which return value would you like to have? The raw ADC reading? Or a value in millivolts? Maybe a function for both...

I'm actually working on the ADC myself at the moment because we experienced some inaccuracies (maybe caused by a bug in the calibration). So I might implement it straight away.

luizvilla, 2021-02-11:
At the moment we are trying to get raw values to start our development chain. Values in milivolts would be interesting as well, but maybe they should be thought within the context of a wider vision. Right now we have structures that hold values, device tree definitions that hold multpliers/dividers and all of that is put together through .c and .cpp drivers. I think (it's only my opinion) that the overall architecture of the code is too complex. 
Ideally we should aim for something like: 
- Measurement
- Decision (Algorithm)
- Action (PWM and others)

All the base measurement functions should be easily available and all the base action functions as well. That's ADC and HRTimer right now. 
More complex measurements (i.e. updating values to milivolts or putting them within a table of sorts) should be a second class of API. More complex actions are GPIO based (activating a mosfet for measurement) or a DAC (for measuring bi-directional current or using HRTimer advanced current functions). 
Algorithms are very specific to the application, but there are basically two types: DC reference tracking or AC reference synchronization. In either case, we'll need a PID to be tuned to the parameters of the system. So, in algorithmic terms, we should aim for specific self-tuning functions coupled with generic PID (CMSIS) functions. 
We are working towards this genericity based goal. I know it is different from the work at LibreSolar, but I do think you guys could benefit from a more simplified and generic code for creating future applications and educating people in terms of power electronics.

hugueslarrive, 2021-02-11:
> Which return value would you like to have? The raw ADC reading? Or a value in millivolts? Maybe a function for both...
> 
Both, I think it would take a function that returns a raw value in daq_driver.c, and functions that convert those values in daq.cpp...