LibreSolar/charge-controller-firmware #118: Missing documentation on how to change battery configuration if using Platformio

hogthrob, 2021-04-04:
I was able to build firmware using from master and develop branches using PlatformIO and provided instructions. 

However, I was not able to find out by reading the documentation if changing the default battery configuration is possible using PlatformIO tooling (in VSCode) at all and what would be the approach.  Maybe it just works with the native zephyr tool set. As defining the battery type/information is one of the most common adjustments, it should be documented how to do this right.

martinjaeger, 2021-04-06:
True, the documentation should be improved here. Thanks for raising the issue!

One way is to adjust the setpoints manually via ThingSet protocol. They are afterwards stored in the EEPROM or flash memory.

For a change at compile-time you can for example add the following to `zephyr/prj.conf` or create a board-specific configuration like `zephyr/boards/mppt_1210_hus.conf`:
```
CONFIG_BAT_TYPE_NMC=y
CONFIG_BAT_NUM_CELLS=3
```

The different types are defined in `zephyr/Kconfig` file.

If using the Zephyr toolchain, you can access the settings via `west build -t menuconfig`. Unfortunately, this is not supported by PlatoformIO.

martinjaeger, 2021-04-14:
This has been fixed with today's release. Please see Readme.md for documentation and comments in `prj.conf`.

If you think that further improvements are necessary, please re-open.

closed 2021-04-14 by martinjaeger