LibreSolar/charge-controller-firmware #126: MPPT 2420 LC "No battery temp sensor" detection does not work anymore on v0.21

hogthrob, 2022-01-16:
The calculated "open sensor" temperature on the MPP 2420 LC is around -32.5 and not below -50 as required for 

https://github.com/LibreSolar/charge-controller-firmware/blob/a56b0d0e8046fcb911dd2a21015845509d13565a/app/src/daq.cpp#L251

to do its magic. Was working with old firmware from 2019. Could not see the difference in the code easily, looks to me more or less identical, hence raising the issue here. I personally solved the problem by entirely disabling the sensor via an `mppt_2420_lc.overlay` file in the app directory:
```
/*
 * Copyright (c) The Libre Solar Project Contributors
 *
 * SPDX-License-Identifier: Apache-2.0
 */

/ {
        pcb {
                version-str = "v0.10";
                version-num = <10>;
        };

        adc-inputs {
                // /delete-node/ temp-bat;
        };
};
```
but it would be nicer, if we get back the original automatic detection capability.

martinjaeger, 2022-01-16:
That's interesting. Previously I got values below -70°C IIRC, so I thought the -50 setting would be pretty safe. I'll double-check if the Zephyr pinctrl driver set some pull-ups or pull-downs automatically, which would be an explanation for the changed behavior.

hogthrob, 2022-01-16:
Just for the record: I tried to put a 10k resistor on the NTC connections and it does show exactly the expected 25°C on the display.